WeActStudio/WeActStudio.MiniSTM32F4x1 #1: Are the 2 CC pins on the USB-C Connector shorted and sharing R8 (5.1kOhm) to GND?

thinxer, 2020-01-19:
I'm having problem connecting the board to MacBook Pro with the default USB-C to USB-C cable, which has a E-Mark chip. The board is not powering on with E-Mark cable. It is working well with a USB-C cable without E-Mark, or a USB-A to USB-C cable.

Looking at the V20 Schematics, I noticed that the 2 CC pins are shorted. The board should have a 5.1kOhm pulldown resistor on each of the CC1 and CC2 lines, not shorting them.

It seems that Raspberry Pi 4B has made a similar problem. Here is a in-depth analysis: https://hackaday.com/2019/07/16/exploring-the-raspberry-pi-4-usb-c-issue-in-depth/.

WeActTC, 2020-01-20:
Hi, because most USB-C cables do not have E-Mark chips, and this situation was not considered at the time, the board can work well with USB-C cables without E-Mark. Thanks for your feedback. We will improve in the next batch of products, each CC will add 5.1kΩ resistance.

thinxer, 2020-01-20:
Thanks for the response. I believe it should be better to leave this issue open so that we can track the progress. People experiencing the same issue thus could subscribe to this issue and get notified when the new batch is ready.

samveen, 2020-02-05:
@thinxer I noticed that V2.1 has one of the CC pins grounded via 5.1kOhm resister (R9), but not the other. Would you review: https://github.com/WeActTC/MiniF4-STM32F4x1/blob/master/STM32F401CCU6-Black-Board/HDK/MiniF4x1Cx_V21%20SchDoc.pdf

thinxer, 2020-02-05:
@samveen The two CC pins  are using the same PIN name and thus shorted, not that one of them is floating.

kamilion, 2020-02-20:
There needs to be two 5.1k ohm resistors to make e-marked cables work properly. 

One for CC1, one for CC2. 

Trying to share one resistor by linking both CC1 & CC2 together through the same resistor is what causes the E-mark problem.

This causes enumeration of the device as a passive headphone adapter that requires no power, and as such, none is delivered.

Also, the v2.1 board has another problem: Pressing Boot0 and Key at the same time causes a deadshort between 3.3v and GND. Another resistor like R6 added between key and ground will solve this current from becoming too large and causing the board to burn up due to unintended deadshort.

thinxer, 2020-03-02:
The issue has been resolved in V2.2 by e6cc12705.

closed 2020-03-02 by thinxer