WeActStudio/WeActStudio.MiniSTM32F4x1 #2: External power supply + USB at the same time

flatpedals, 2020-01-28:
Regarding the 411CE v1.3, is there any protection in place that would allow connecting USB and an external 5V source at the same time? Is this recommmended? 
Thank you!

WeActTC, 2020-01-28:
Hello, it is not recommended to use external power supply and usb at the same time, because there is no diode protection. Common ground is fine, or connect a diode in a `5V Pin` to prevent current backflow to the usb.We will make corresponding improvements in the next hardware version

cgo, 2021-01-13:
Does this problem also persist in the 3.0 board with F401CEU?

flatpedals, 2021-01-13:
afaik, v3 has a diode in place for this specific issue. I had a v3 F411 for a short while and the diode was there but I didn't test. Maybe check the diagrams, but I think it has been fixed.

cgo, 2021-01-13:
Thanks @flatpedals