WeActStudio/WeActStudio.MiniSTM32F4x1 #17: USB HID not working on F411CEU6 using Micropython

womei, 2020-11-18:
When trying to send USB HID commands, the controller reboots.

See [This live stream for real time example]( https://www.reddit.com/rpan/r/shortcircuit/jvum89)

Now, for more details on the issue. I am using a board (with no extra ram) running the latest official image

Trying to use HID via either the official [micropython tutorial](https://docs.micropython.org/en/latest/pyboard/tutorial/usb_mouse.html?highlight=hid) or [this tutorial](http://wiki.micropython.org/USB-HID-Keyboard-mode-example-a-password-dongle) causes the controller to crash when the `hid.send()` function is called.

DejayRezme, 2021-01-06:
I'm a noob looking to maybe use this board for a keyboard / trackball combo. 
Is this issue only for micro python or also for C++ library?
Has it been resolved? Thanks

DejayRezme, 2021-01-06:
I have no clue, but I stumbled across this, maybe related? 
https://github.com/qmk/qmk_firmware/tree/master/keyboards/handwired/onekey/blackpill_f411

womei, 2021-01-06:
This is just for micropython, I have not tried with the C/C++ libraries.