WeActStudio/WeActStudio.MiniSTM32F4x1 #25: 5V measured on the 3.3V pins

szszoke, 2021-04-13:
Hello!

I got two Blackpill V3 boards with the F411CE and when they are connected via the USB cable to my computer, I can measure 5V on both the 5V and the 3.3V pins.

I checked and they are not shorted. Is this normal?

szszoke, 2021-04-13:
If I am reading the schematic correctly, the 3.3V pins should be connected to the output of the 3.3V regulator.

WeActTC, 2021-04-13:
I think the LDO failed, maybe your board is not made by us, it is one of the copy board, this board, if you short circuit 3v3 and GND, LDO will fail, it can not output 3.3V, instead of output 5V. The worst part is that it printed our version number and logo, and placed a Github link on AliExpress's product details page

szszoke, 2021-04-14:
I have two boards, new, and they both exhibit this issue.

szszoke, 2021-04-14:
Where is the LDO located on the board? I'm guessing it is U2 on the schematic but it is not marked on the board.

szszoke, 2021-04-15:
I ended up doing more testing and I found that the boards have a 5V regulator instead of a 3.3V regulator.

closed 2021-04-15 by szszoke

Dreadrik, 2022-01-23:
> I ended up doing more testing and I found that the boards have a 5V regulator instead of a 3.3V regulator.

I just received two boards with the exact same issue. Bought them from a reputable seller as well. Did you manage to replace the regulator, or did you return the boards?

szszoke, 2022-01-23:
I still have the old boards but I got replacements. The replacement boards had really straight edges on the PCB while the bad ones looked like they were marked with a saw and then broken off.

Dreadrik, 2022-01-23:
Ok, thanks! Mine look straight, has the WeAct logo and the V3.0 version number and everything. I think I'll have to make a claim for replacements as well.

fivdi, 2022-01-23:
The version number of the latest boards has been V3.1 for quite some time now.

See https://www.aliexpress.com/item/1005001456186625.html.
See also https://github.com/WeActTC/MiniSTM32F4x1/tree/master/HDK.