WeActStudio/WeActStudio.MiniSTM32F4x1 #26: Flashing via ST-Link V3

szszoke, 2021-04-13:
Hello!

I have two of the V3 boards with the F411CE and I am unable to make my ST-Link V3 detect the chips.

My ST-Link is running the `V3J7M3B5S1` firmware, which as far as I know is that latest one.

I am trying to program via SWD.

Here are my connections
-------------------
| ST-Link V3 | Board |
| ---------- | --------- |
| GND | GND |
| VCC | 3V3 |
| CLK | SCK |
| DIO | DIO |

The board is plugged in via USB and is running an Arduino blink example, which was uploaded via DFU mode.

I made sure with a multimeter that my connections are good.

```
st-info --probe
Found 1 stlink programmers
 serial:     303033413030314533303339353130383334333933383338
 hla-serial: "\x30\x30\x33\x41\x30\x30\x31\x45\x33\x30\x33\x39\x35\x31\x30\x38\x33\x34\x33\x39\x33\x38\x33\x38"
 flash:      0 (pagesize: 0)
 sram:       0
 chipid:     0x03e8
```

```
openocd -f interface/stlink.cfg -f target/stm32f4x.cfg                                                                                                                                                                                                                            
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V3J7M3B5S1 (API v3) VID:PID 0483:374F
Info : Target voltage: 4.638830
Error: init mode failed (unable to connect to the target)
```

STM32CubeProgrammer is also failing to connect to the chip:
```
  01:34:00 : STM32CubeProgrammer API v2.7.0
  01:34:16 : ST-LINK SN  : 003A001E3039510834393838
  01:34:16 : ST-LINK FW  : V3J7M3B5S1
  01:34:16 : Board       : STLINK-V3SET
  01:34:16 : Voltage     : 4,60V
  01:34:16 : No STM32 target found!
  01:34:16 : ST-LINK SN  : 003A001E3039510834393838
  01:34:16 : ST-LINK FW  : V3J7M3B5S1
  01:34:16 : Board       : STLINK-V3SET
  01:34:16 : Voltage     : 4,57V
  01:34:16 : Error: No STM32 target found!
```

Holding the reset button while trying to connect to the chip doesn't work either.

I tried this both on Linux and Windows. Is there something that I am missing here?

denizzzka, 2021-04-14:
>openocd -f interface/stlink.cfg -f target/stm32f4x.cfg  

Try to use `target/stm32f1x.cfg` config

szszoke, 2021-04-14:
I tried it but it didn't work.

I think I'm dealing with counterfeit chips here.

The 3.3V line is at 5V, and the STM32 is also getting 5V.

I have two boards from the same store and they are both behaving like this (including 5V at the 3.3V line)

denizzzka, 2021-04-14:
Can you also check your St-link device to ensure that its VCC output it is not source of +5V?

szszoke, 2021-04-15:
Turns out my boards have a 5V regulator instead a 3.3V one, so I'm getting a set of new ones.

I will report back when they arrive.

denizzzka, 2021-04-16:
> Turns out my boards have a 5V regulator instead a 3.3V one, so I'm getting a set of new ones.

My (unofficial) ST-Link V2 provides 3.3V output pin, maybe V3 also does?

szszoke, 2021-04-19:
As far as I know the official ST-Link programmers do not power the target.

I got my replacement boards which seem to be working and are detected correctly by ST-Link as well.

closed 2021-04-19 by szszoke