WeActStudio/WeActStudio.MiniSTM32F4x1 #28: digitalWrite outputs 3.3V even though the board is 5V feeded

Akrai, 2021-04-15:
So I'm trying to figure this out, I thought the board should output, when using digitalWrite (stm32duino), 5V if the board is 5V powered, and 3.3V if it is 3.3V powered, but I'm getting 3.3V on all the pins I write with the board being 5V powered.

luuke, 2021-04-15:
Microcontroller is supplied with 3.3V and so it outputs such voltage.

Hackswell, 2021-04-15:
I believe that the chip itself is 3V3.  It can accept 3V3 or 5V input, but it's converted into 3V3 for the chip.  If you need the pins to output 5V, you'll need to step it up with a converter.  (I have to do this with my NeoPixel LED strips)

Akrai, 2021-04-15:
But for example, with an arduino nano it behaves like i want, 5V if it is 5V powered, but the chip on the nano is 3.3V powered, am i right?

Are other STM32 boards behaving like this?

luuke, 2021-04-15:
Arduino Nano has ATmega chip which is 5V. It can work with 3.3V. STM32 are 3.3V and applying 5V to it might damage it.

Akrai, 2021-04-15:
Okay, I just want to make very sure that the board won't output on any I/O pin 5V under any circumstance, as I won't be using a digital shifter to protect a TFT screen (as those don't work if input and output is 3.3V) which will break if it gets 5V

Hackswell, 2021-04-15:
Correct.  From everything I can see, INPUTS can be 3V3 or 5V (on most pins... see the chart I made for which pins on F411 aren't 5V tolerant).  OUTPUTS are 3V3.   

[https://github.com/WeActTC/MiniSTM32F4x1/blob/master/images/STM32F4x1_PinoutDiagram_RichardBalint.png](url)

closed 2021-04-15 by Akrai