WeActStudio/WeActStudio.MiniSTM32F4x1 #35: Is it possible to program the WeActV3.0 with STM32CubeIDE?

dddomodossola, 2022-05-08:
Hello,

I would like to program the Weact v3.0 with STM32CubeIDE. 
Do you provide a .ioc file? Or which board to select?

Kind Regards.

zoggn, 2022-05-08:
Of course it is possible. Just select stm32f411 or f401.

dddomodossola, 2022-05-08:
Thank you for the fast reply.
But, this way I have to configure hardware manually. Doesn't it?

zoggn, 2022-05-08:
> Thank you for the fast reply. But, this way I have to configure hardware manually. Doesn't it?
 
You do. That's an development board, not a ready device. So, you could configure it whatever you want for your needs

dddomodossola, 2022-05-08:
:-D That's true

closed 2022-05-08 by dddomodossola

cshawky, 2022-05-08:
Hi David. 
The hour or so you spend with STMCubeMX (not STM32CubeIDE) to create your own IOC file is well worth it. Consider it a training session. If you are already ofay with MX then it should only take about 5 minutes to mock up any of the WeAct devices. I have been doing exactly that as part of my learning curve.
When you use MX directly it effectively gives you the opportunity to create a separate MX project which then becomes a template for the IDE, that you can re use.
I suggest you start with the latest Schematic to do your pin assignments.

dddomodossola, 2022-05-09:
@cshawky thank you. I preferred to not to use Arduino IDE because it is a really poor environment. 
Furthermore I want to learn ST sdk because I will start a commercial project in industrial environment, and I want complete understanding about the device.

cshawky, 2022-05-09:
Same same. I am having a look at Arduino, Block coding support etc as one solution I am putting together would be ideal if the client could maintain his control logic. The client is not a programmer. My son and I prototyped his requirement using Scratch! Middle ground might be the Arduino IDE, if the embedded block code ends up being no good (I'm not seeing recent updates in that space).
I started with the STM32VLDiscovery and grabbed two Black Pills online (eBay) as the VL ST-LINK V1 interface is no longer supported. Unfortunately or fortunately I have returned the Black Pills to the vendor, not fit for purpose and order two from WeAct directly (almost the same price) once I realised potential copying.
This is off topic but the first unit worked fine via ST-LINK V2. I plugged in USB-C once but never used the USB interface. After a weekend of developing, I went to use the second unit but it did not power up. No 3.3V rail. Power from USB was 4.98V. So I plugged USB-C into the unit I had been using the ST-LINK, and got 4.8Vdc on the 5V rail and 4.5Vdc on the 3.3V rail. By the time I had finished videoing and comparing with a multimeter to put a case for replacement the first unit also died. i.e. 4.5V fried it.
Long story short the little voltage regulator wasn't any good. My concern now is that the genuine Mini from WeAct will be no different as for the life of me I could not pick any difference. The units I had were stamped WeAct 3.0 and of good build quality.

cshawky, 2022-05-15:
I have uploaded my IOC definition for this board to https://github.com/cshawky/STM32CubeMX

dddomodossola, 2022-05-15:
@cshawky Thank you a lot ;-)

cshawky, 2022-05-16:
I ran out of time to post my sample app which I feel would be helpful for PWM setup. Please post a comment on my repository in the discussions section to give feedback for others to benefit. More than happy to hear how you fared. I am just a novice with the STM32 kit.