WeActStudio/WeActStudio.MiniSTM32F4x1 #45: Noisy 3.3v (is it a clone?)

bobemoe, 2023-10-10:
We are prototyping a big project using these and have probably used nearly 100 so far from different suppliers. We find some are having inaccurate ADC reading. We have traced this down to what we suspect is a noisy 3.3v supply. See below for some oscilloscope shots of the 3.3v rail on a good, ok and bad board respectively. All tests powered from the same stable 5v USB.

1. I suspect some are probably clones. Is there a definite way to check if they are clones or not?
2. Has anyone else seen this kind of noise on the 3.3 rail or could this issue be something else?
3. We may try replacing the regulator, what is the official regulator used on the genuine boards?

Good (Stable ADC readings)
[image]

OK (Slight errors in ADC)
[image]

Bad (Wildly off ADC)
[image]

WeActStudio, 2023-10-13:
Please check whether there is WeAct Studio V3.1 on the back of the board.

bobemoe, 2023-10-13:
We have a very varied selection of v3.1, v3.0 some with WeAct, some with WeAct Studio, some without and some blank! Many combinations!

I created a simple test code to measure the internal avref over 10 seconds and print the standard deviation.

All of the 3.1's with the official logo were the most stable.
The "bad" ones were all "WeAct v3.0" but also a lot of this type were good too.
All the others were acceptable but not as good as the 3.1's.

Think we must just have a lot of fakes. We want to buy a batch of official devices the link to official shop in the README is broken see https://github.com/WeActStudio/WeActStudio.MiniSTM32F4x1/issues/42

bobemoe, 2023-10-13:
Here is photos of all the devices I tested. 
The black number is the test result (mV range of AVREF over 10secs)
The white number is our test id ref.

## BAD ( > 35mV )
[image]

## GOOD ( < 35mV )
[image]

WeActStudio, 2023-10-14:
The v3.0 you bought is fake. Now the boards we sell are all v3.1. This is a AliExpress link weactstudio.aliexpress.com