WeActStudio/WeActStudio.MiniSTM32F4x1 #47: Datasheet for STM32 F405 and F446

synersignart, 2024-02-02:
Hi would need explanation on these two board the STM32F405 and STM32F446 as there are very little documentation and schematics found online
particular on the SB jumper and the SD card  SDIO or SPI information

WeActStudio, 2024-02-02:
https://github.com/WeActStudio/WeActStudio.STM32F4_64Pin_CoreBoard.git

---- Replied Message ----
| From | Cliff ***@***.***> |
| Date | 02/02/2024 10:34 |
| To | ***@***.***> |
| Cc | ***@***.***> |
| Subject | [WeActStudio/WeActStudio.MiniSTM32F4x1] Datasheet for STM32 F405 and F446 (Issue #47) |

Hi would need explanation on these two board the STM32F405 and STM32F446 as there are very little documentation and schematics found online
particular on the SB jumper and the SD card SDIO or SPI information

synersignart, 2024-02-02:
okie

synersignart, 2024-02-02:
Does the  PA8 needs to be active inorder to operate the TF?